pycom/pycom-micropython-sigfox #487: 'str' object has no attribute 'capitalize'

William04A, 2020-10-14:
**Pycom Board:** GPy 1.0 with Expansion Board 3.1

**uname info (parsed by my code):** SYSTEM INFO: sysname:'GPy', nodename:'GPy', release:'1.20.2.r0', version:'v1.11-783192e on 2020-08-06', machine:'GPy with ESP32', pybytes:'1.5.1'

**Problem:** I have no idea why this happens, but `capitalize()` is not a string function according to the error.

The code:

 ```python
sensor_type = str(sensor["type"]).capitalize() #Sensor dict looks like: {"type": "Temperature", "model": "something"} 
```

The exception:

`AttributeError: 'str' object has no attribute 'capitalize'`

I will probably be able to solve the issue in my code by finding a workaround, such as `string.capwords()`, but I feel like you might want to consider investigating this.

robert-hh, 2020-10-14:
Micropython does not support each and every method of CPython - because it is Micro. As you said, you can easily rebuild the capitalize() with an own function. You can also make your own class built on the str class which supports capitalize.
e.g.
```
class mystr(str):
    def __init__(self, s):
        self.s = s
    
    def capitalize(self):
        if len(self.s) > 1:
            return self.s[0].upper() + self.s[1:]
        else:
            return ""
```

William04A, 2020-10-14:
@robert-hh The function used to work, which was what I found strange.

But if this is intended, I'll close the issue.

closed 2020-10-14 by William04A

robert-hh, 2020-10-14:
> The function used to work.

I cannot remember ever have seen this method being implemented in the Pycom or MicroPython.org versions. It's also not in the Circuitpython, Digi and SiPeed variants.

William04A, 2020-10-14:
> > The function used to work.
> 
> I cannot remember ever have seen this method being implemented in the Pycom or MicroPython.org versions. It's also not in the Circuitpython, Digi and SiPeed variants.

My bad then. I never got an error even though the function was run on every run, it first appeared when I updated the device.

But thanks, I probably just confused it.