pycom/pycom-micropython-sigfox #490: LoRa crashes immediatel

ichatz, 2020-10-18:
I get the error with both firmwares:
* sysname='LoPy4', nodename='LoPy4', release='1.20.2.rc11', version='v1.11-d574024 on 2020-08-25', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.5.2', pygate='1.0.1'
* sysname='LoPy4', nodename='LoPy4', release='1.20.2.rc10', version='v1.11-b2436d4 on 2020-07-01', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.5.0', pygate='1.0.1'

I initialize the LoRa module using the init command.
* When no region is passed, a "ValueError: no region specified and no default found in config block" is given and the initialization aborts.
* When a region is passed, I get a core dump.

```
from network import LoRa
import socket
import binascii
import struct
import time

lora = LoRa(mode=LoRa.LORAWAN, region=LoRa.EU868)
```

I get the same behavior with LoRa.LORA.

Gives the same error with all the examples for LoRa from the pycom-libraries repository.

Thymo-, 2020-10-28:
Does the coredump provide any additional information?

robert-hh, 2020-10-28:
Yes. The code line ang register values of the crash.