pycom/pycom-micropython-sigfox #492: Incompatible IDF git hash, LOPY4

gsbwallace1996, 2020-10-29:
Error Message for command 'make BOARD=LOPY4':
```
fatal: not a git repository: '/home/gsbwallace1996/pycom-esp-idf/.git'

Incompatible IDF git hash:

c61fe64 is expected from IDF_HASH from Makefile, but
 is what IDF_PATH=/home/gsbwallace1996/pycom-esp-idf is pointing at.

You should probably update one (or multiple) of:
  * IDF_PATH environment variable
  * IDF_HASH variable in esp32/Makefile
  * IDF commit, e.g.
cd $IDF_PATH && git checkout c61fe64 && git submodule sync && git submodule update --init --recursive && cd -

make: *** [application.mk:921: CHECK_DEP] Error 1

```
I have been attempting to build the LOPY4 firmware using both the Windows MSYS terminal and the Ubuntu terminal.
For both I get the same error which is similar to: https://github.com/pycom/pycom-micropython-sigfox/issues/464
However, unlike the above link, my error is not pointing to any IDF_HASH. The mpy-cross built successfully.

Any help would be much appreciated.

peter-pycom, 2020-10-29:
can you show the output of the following commands from Ubuntu
```
echo $IDF_PATH
ls -al $IDF_PATH 
cd $IDF_PATH
git status
git rev-parse --short HEAD
git -c core.abbrev=7 --git-dir=$IDF_PATH/.git rev-parse --short HEAD
```

gsbwallace1996, 2020-10-29:
Problem fixed, IDF_PATH incorrect:
`export IDF_PATH=~/pycom-esp-idf`
failed to account for an esp32 folder.
Thanks for the help

closed 2020-10-29 by gsbwallace1996