pycom/pycom-micropython-sigfox #496: LoPy4 RED LED after OTA method B firmware flash (meaning?)

danalvarez, 2020-11-13:
Hello all:

I am using a LoPy4 with firmware version 1.20.0.rc13.

I am updating the firmware version from 1.20.0.rc13 to 1.20.2.rc6, via FileZilla (FTP) using the method B described here: https://docs.pycom.io/updatefirmware/ota/

However, this is what happens:

1. Upon sucessful transfer of `appimg.bin` to the `/flash/sys/` directory, I then reset the LoPy4.
2. When the LoPy4 boots again, it shows **a RED LED** for 2 or 3 seconds, and then moves to the standard blue heartbeat.
3. I can then no longer access the LoPy4 via the UART0 pins, and have to performe a **safe boot** to then perform a full flash erase via `pycom-fwtool-cli` and then manually flash the firmware via the CLI.
4. When the firmware (v1.20.2.rc6) is manually flashed, the LoPy no longer exhibits the RED LED prior to the standard heartbeat and all is well again.

What happens:

1. Firmware flashing via OTA method B results in what I presume is a corrupted flash, and the LoPy4 no longer communicates via serial.
2. Additionally, the LoPy4 exhibits a RED LED for 2 or 3 seconds after boot when in this state.

What should happen:

1. After flashing the firmware via OTA method B and resetting, the LoPy4 should fall back into the standard hearbeat with no prior LED color. And should be able to communicate via REPL on UART0.

Any ideas, guys? What does the RED LED mean?

Thank  you in advance,

Dan

gijsio, 2020-11-13:
Hi,
I had exactly this case today coincidentally. The red LED means there is some error in the boot process while loading all variables and configuration.
You could erase either the CONFIG and/or NVS partitions on the device to resolve the issue. Im not exactly sure what causes it, but it seems to be related to some corruption/inconsistency in either of those partitions making the RGB LED turn red for 3-4 seconds.
Let me know if that works for you

danalvarez, 2020-11-13:
Hello, how would I go about erasing the config partition? And do you mean erasing the NVS partition by running `pycom.nvs_erase_all()`?

My main issue with this is that when this happens, I must remove the LoPy4 from the enclosure it is currently in, to perform a safe boot and get it out of this broken state. Do you know of any way of solving this that does not involve safe boot?

My main problem is that, when in this state, the REPL seems not to initialize, as I am unable to communicate via UART0. The LoPy4 boots, but the `>>>` do not come up in the terminal, so I cannot talk to it...

gijsneerhof, 2020-11-15:
You can clear the CONFIG partition with the pycom firmware updater tool (theres probably other methods I do not know about), this will not need safeboot, only a USB connection

danalvarez, 2020-11-17:
Hello, thanks for the response!

However, the disadvantage of using the Pycom Firmware Updater tool still stands, because in order to use it I need to enter bootloader mode, and I cannot do this unless I put P2 to GND.

Is there any other way that does not involve using the Pycom Firmware Updater Tool?

gijsio, 2020-11-20:
You could try the CLI interface to set the CONFIG values: https://docs.pycom.io/advance/cli/ 
Im not sure how far that will get you without using bootloader mode though