pycom/pycom-micropython-sigfox #512: Error waking from sleep when in Wifi STA mode

chriskoz, 2020-12-28:
Getting an OSError when waking from machine.sleep() with resume_wifi_ble=True if the Wifi is in STA mode but NOT connected to anything.

Using WiPy 3.0 with firmware ver. 1.20.2.r2

This simple main.py demonstrates the issue:
```
import machine
from network import WLAN
wlan = WLAN(mode=WLAN.STA)
print('sleep...')
machine.sleep(1000, True)
print('awake.')
```

This results in the following output:
```
sleep...
Traceback (most recent call last):
  File "main.py", line 5, in <module>
OSError: the requested operation failed

```

This problem also occurs if you try to connect to an AP but fail.  For instance because the Wifi AP is offline, out of range, or misconfigured.  (My Wifi was out of range when I discovered this issue)

gijsio, 2021-01-15:
Hi,
I can indeed reproduce this issue! Thanks for informing us.
The issue indeed has to do with not setting any WLAN parameters beforehand through the initialisation, except for the mode. My proposed fix would include a check for the SSID to be set as a requirement in order to resume the WLAN. If there is no SSID, it will not go through `wlan_setup(..)`. Let me know what you think
Gijs

chriskoz, 2021-02-22:
Hi Gijs,

Assuming the SSID check would also handle failed connection attempts as well... then that fix certainly sounds like it would work.

Thanks.