pycom/pycom-micropython-sigfox #518: LFS: Core dumps at file close

robert-hh, 2021-01-20:
Firmware: 1.20.2.r4, WiPy3, with a change for block level wear leveling and RGB led guard.
The error happens during a long term test, aiming at writing the same short file several million times, in order to verify the wear leveling change.
At random intervals, about every 50_000 file open/write/close cycles, a core dump happens. It may happen after a few thousand cycles, or after 100_000 cycles.
Test code:
```
import time
line = "*" * 63 + "\n"
b = 31 * line
        
def write(name, n):
    f = open(name, "w")
    f.write("%8d\n" % n)
    f.write(b)
    f.close()

def run1(start=1, end=10_000_000):
    try:
        f = open("logfile")
        l = f.readline()
        f.close()
        n = int(l.strip())
    except:
        n = start
    print(n, "Restart ")

    while n < end:
        print(n, "--- ")
        write("logfile", n)
        n += 1
        time.sleep(0.1)
```
The backtrace of a typical core dump looks like. I decoded 6 core dumps. They all happen on a file close. And the exception always happens at the rmt_tx interrupt, which is served by Core1. RMT is activated by the heartbeat in the main loop, which is served by Core 0. The backtrace of Core shows, that at the time of the heartbeat the core 0code is working on sending a RGB led flash. Whether still or again, is unclear.
```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4009602e  PS      : 0x00060034  A0      : 0x800952ec  A1      : 0x3ffbf4f0  
A2      : 0x3ffe2b4c  A3      : 0x00060023  A4      : 0x00000001  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x3ffdd385  A8      : 0x0000cdcd  A9      : 0x00000000  
A10     : 0x0000039f  A11     : 0x0000abab  A12     : 0x3f40df58  A13     : 0x00000000  
A14     : 0x3f40df40  A15     : 0x01f12318  SAR     : 0x00000003  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40093a80  LEND    : 0x40093a93  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400f5227  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4009602e

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4009602e:0x3ffbf4f0 0x400952e9:0x3ffbf520 0x40088749:0x3ffbf540 0x40084b3d:0x3ffbf580 0x400f5224:0x3ffdd3b0 0x400f557a:0x3ffdd3e0 0x400f7375:0x3ffdd460 0x400f7a35:0x3ffdd520 0x400f7dc9:0x3ffdd5d0 0x400f7df5:0x3ffdd610 0x400f89c1:0x3ffdd630 0x400f9672:0x3ffdd650 0x400f9738:0x3ffdd670 0x4010c991:0x3ffdd690 0x40107759:0x3ffdd6c0 0x40103b61:0x3ffdd6e0 0x40103bf1:0x3ffdd700 0x40110a57:0x3ffdd720 0x4010786c:0x3ffdd7c0 0x40103b61:0x3ffdd830 0x401109c9:0x3ffdd850 0x4010786c:0x3ffdd8f0 0x40103b61:0x3ffdd920 0x40103bf1:0x3ffdd940 0x40110a57:0x3ffdd960 0x4010786c:0x3ffdda00 0x40103b61:0x3ffdda80 0x40103b8e:0x3ffddaa0 0x400e6f15:0x3ffddac0 0x400e71a5:0x3ffddb60 0x400e5dd3:0x3ffddb80

Core 0 register dump:
PC      : 0x40097d80  PS      : 0x00060c34  A0      : 0x80096b48  A1      : 0x3ffbb960  
A2      : 0x3ffe2b24  A3      : 0x3ffbbb28  A4      : 0x3ffbbb0c  A5      : 0x00000001  
A6      : 0x000000fe  A7      : 0x00060023  A8      : 0x3ffbbb28  A9      : 0x00000019  
A10     : 0x3ffbbb28  A11     : 0x00000019  A12     : 0xb33fffff  A13     : 0x0000cdcd  
A14     : 0x000000fe  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400f27a8  LEND    : 0x400f27eb  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40097d80:0x3ffbb960 0x40096b45:0x3ffbb980 0x40095473:0x3ffbb9a0 0x401690f1:0x3ffbb9e0 0x400f27fa:0x3ffbba10 0x400f9bfe:0x3ffbba30 0x400d3cf1:0x3ffbba50 0x400967bd:0x3ffbba70

------------------------------------------------
Core 1 Backtrace
--------------------------------------------

BT-0: 0x4009602e is in vTaskEnterCritical (/home/peter/docs/pycom-esp-idf/components/freertos/include/freertos/portmacro.h:355).
355	    __asm__ __volatile__ (
BT-1: 0x400952e9 is in xQueueGiveFromISR (/home/peter/docs/pycom-esp-idf/components/freertos/queue.c:1324).
1324			taskENTER_CRITICAL_ISR(&pxQueue->mux);
BT-2: 0x40088749 is in rmt_driver_isr_default (/home/peter/docs/pycom-esp-idf/components/driver/rmt.c:579).
579	                        xSemaphoreGiveFromISR(p_rmt->tx_sem, &HPTaskAwoken);
[... 53 lines cut ...]
560	    return parse_compile_execute(filename, MP_PARSE_FILE_INPUT, EXEC_FLAG_SOURCE_IS_FILENAME);
BT-30: 0x400e5dd3 is in TASK_Micropython (mptask.c:339).
339	            int ret = pyexec_file(main_py);
---------------------------------------------
Core 0 backtrace
---------------------------------------------
reading backtrace from stdin
0x40097d80:0x3ffbb960 0x40096b45:0x3ffbb980 0x40095473:0x3ffbb9a0 0x401690f1:0x3ffbb9e0 0x400f27fa:0x3ffbba10 0x400f9bfe:0x3ffbba30 0x400d3cf1:0x3ffbba50 0x400967bd:0x3ffbba70
BT-0: 0x40097d80 is in vListInsert (/home/peter/docs/pycom-esp-idf/components/freertos/list.c:162).
162			pxIterator = pxList->xListEnd.pxPrevious;
BT-1: 0x40096b45 is in vTaskPlaceOnEventList (/home/peter/docs/pycom-esp-idf/components/freertos/tasks.c:2901).
2901		vListInsert( pxEventList, &( pxCurrentTCB[ xPortGetCoreID() ]->xEventListItem ) );
BT-2: 0x40095473 is in xQueueGenericReceive (/home/peter/docs/pycom-esp-idf/components/freertos/queue.c:1590).
1590					vTaskPlaceOnEventList( &( pxQueue->xTasksWaitingToReceive ), xTicksToWait );
BT-3: 0x401690f1 is in rmt_write_items (/home/peter/docs/pycom-esp-idf/components/driver/rmt.c:819).
819	    xSemaphoreTake(p_rmt->tx_sem, portMAX_DELAY);
BT-4: 0x400f27fa is in led_set_color (mods/modled.c:109).
109	    for (uint32_t bit_mask = MSB_VALUE; bit_mask != 0; bit_mask >>= 1){
BT-5: 0x400f9bfe is in mperror_heartbeat_signal (util/mperror.c:142).
142	                led_set_color(&led_info, false, false);
BT-6: 0x400d3cf1 is in esp_vApplicationIdleHook (/home/robert/Downloads/MicroPython/pycom-esp-idf/components/esp32/freertos_hooks.c:51).
51	        if (idle_cb[core][n] != NULL && !idle_cb[core][n]()) {
BT-7: 0x400967bd is in prvIdleTask (/home/peter/docs/pycom-esp-idf/components/freertos/tasks.c:3382).
3382				vApplicationIdleHook();
```

robert-hh, 2021-01-24:
Continuing that test:
- WiPy3: After disabling heartbeat, it continued until now for >1 million cycles. I'll let it run, since the test is for flash not wearing.
- FiPy: First core dump after 137000 cycles. w/o heartbeat no problems
- LoPy4: > 1 million cycles WITH heartbeat enabled. The test runs on LoPy4 noticeable faster than on FiPy or WiPy. maybe a different type of flash chip.

robert-hh, 2021-01-26:
The core dump is related to the heartbeat RGB led operation, and caused by a race conditions between switching the LED on and off. On heave flash use, the "On" cycle seems to be delayed long enough to collide with the "Off" cycle. The most simple cure seems to be to change lines 136 and 142 from:
`led_set_color(&led_info, false, false);`
to 
`led_set_color(&led_info, true, false);`
With the change, the new led setting is not executed before the previous one is finished. That test is running at the moment; Since the crash is relatively rare, it will take another 24 hours to get sufficient confidence in the result. Another test I did was simply extending the "On" phase from 80ms to 200ms, which was also fine over a period of 24 hours.

robert-hh, 2021-01-29:
Another small code change made it run now for 48 hours and about 1 million cycles. Even if I am not 100% confident that the change if fixing the core reason for the fault, it makes it not happen. The change is again in mperror.c to the function mperror_heartbeat_signal(). I'll make a PR request for that on Monday, after letting the test run for another ~48 hours.
```
bool mperror_heartbeat_signal (void) {
    if (mperror_heart_beat.do_disable) {
        mperror_heart_beat.do_disable = false;
    } else if (mperror_heart_beat.enabled) {
        if (!mperror_heart_beat.beating) {
            if (mp_hal_ticks_ms_non_blocking() - mperror_heart_beat.off_time > MPERROR_HEARTBEAT_OFF_MS) {
                led_info.color.value = MPERROR_HEARTBEAT_COLOR;
                led_set_color(&led_info, true, false);
                mperror_heart_beat.beating = true;
                mperror_heart_beat.on_time = mp_hal_ticks_ms_non_blocking();
            }
        } else {
            if (mp_hal_ticks_ms_non_blocking() - mperror_heart_beat.on_time > MPERROR_HEARTBEAT_ON_MS) {
                led_info.color.value = 0;
                led_set_color(&led_info, true, false);
                mperror_heart_beat.beating = false;
                mperror_heart_beat.off_time = mp_hal_ticks_ms_non_blocking();
            }
        }
    }
    // let the CPU save some power
    return true;
}
```

geza-pycom, 2021-03-28:
@robert-hh : May I ask that did this last change solve the problem?

robert-hh, 2021-03-28:
@geza-pycom 
> May I ask that did this last change solve the problem?

Yes, it did.

geza-pycom, 2021-04-10:
Thanks, I am closing this issue as the corresponding PR will be part of an upcoming release.

closed 2021-04-10 by geza-pycom