pycom/pycom-micropython-sigfox #519: LTE: `lte_check_attached` returns false

sven-hm, 2021-01-21:
`CEREG` was set to `1` in this [commit](https://github.com/pycom/pycom-micropython-sigfox/pull/499/commits/8e631efefb6b3d7be6731f826c06dcaf85231785), so the answer to an `"AT+CEREG?"` will only contain information up to the  `<stat>`-value. Reading the code of [lte_check_attached in `modlte.c`](https://github.com/pycom/pycom-micropython-sigfox/blob/ffb0e1cf4b34342a2aa689189d3cef732dacaf44/esp32/mods/modlte.c#L255) it seems with `legacy=true` and when `"AT+CGATT?"` returns `0` it will always return false.

I'm wondering if this is expected behavior and if i want to fix it [this part of the conditional](https://github.com/pycom/pycom-micropython-sigfox/blob/ffb0e1cf4b34342a2aa689189d3cef732dacaf44/esp32/mods/modlte.c#L284) can safely be removed?

* I use a Gpy
* It concerns every firmware versions since v1.20.2.r2

gijsio, 2021-01-29:
You're correct this is an issue, you'll probably see it fixed in the next release!

peter-pycom, 2021-02-19:
internal PR: https://github.com/pycom/Firmware-Development/pull/221

scottbasgaard, 2022-06-07:
> internal PR: https://github.com/pycom/Firmware-Development/pull/221

Was this fixed?