pycom/pycom-micropython-sigfox #521: Pymakr crashes when I try to upload a file

mc2software, 2021-01-22:
Using VSCode 1.52.1 and Node.js 12.14.1. Working with a ESP32s board. Whenever I try to upload files to the device, Pymakr crashes and exits the terminal with error = 1.  This happens when I try to upload just one file.  I cannot upload multiple files because, when I right-click on a folder and choose to upload project, it starts copying from the root folder. I've made that mistake twice and had to manually cleanup the folders copied from the prompt.

mc2software, 2021-01-22:
Here's a pic of the issue. Happens every time.
[image]

gijsio, 2021-01-29:
Thanks, we're working on it!