pycom/pycom-micropython-sigfox #527: Continuous boot problem with PIR, lora and deepsleep project.

krukle, 2021-02-02:
(sysname='LoPy4', nodename='LoPy4', release='1.20.2.r3', version='v1.11-044dfea on 2020-12-23', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.6.1')

We've built an application that wakes up from high/lows/ from a HC SR501 PIR sensor. 

First it runs a wifi-sniffer

Then it sends a message thorugh LoRa.

Then it goes to deepsleep, set to wake up from either high or low depending on what it woke up from last. 

It goes smoothly a few cycles. Sometimes even days. But after a while we always run into this problem 

´´´python
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xbfff009c,len:1228
load:0x3f720080,len:1073782456
ets Jun 8 2016 00:22:57
´´´

This loops indefinitely until the power is cut. We can not catch it with an error handler because it wont even let the pycom load boot nor main.

What's happening here?

We've tried on three different pycom lopy4 units and with different firmware's, with lora.nvram.save/load and without. If we dont use Lora, this never happens. And if we dont use deepsleeps, this never happens either (obviously).

robert-hh, 2021-02-02:
Does the device boot again properly after a power cut, or do you have to set it up again?

krukle, 2021-02-02:
It boots up properly after a power cut.

robert-hh, 2021-02-02:
This section looks weird:
```
load:0x3fff0018,len:4
load:0xbfff009c,len:1228
load:0x3f720080,len:1073782456
```
That is output from the boot loader. Comparing to a LoPy4 boot message here and surely to the 'good' messages at your side, addresses and lengths are off, especially the  address in the second line and the length on the last line. The length value 1073782456 is hex 40009eb8, which is a valid address in the internal ROM. 

Can you exclude a problem with the power supply?

krukle, 2021-02-03:
We've used different power supplies and still gotten the same error
message. We've gotten the message through a power bank and through the wall
outlet, and with different cables.