pycom/pycom-micropython-sigfox #529: VSCode Pymakr  command 'pymakr.upload' not found

antifm, 2021-02-04:
[image]
[image]

Recently VScode has updated to version 1.53 and using the Pymakr extension v1.1.7 no longer works. Pymakr no longer starts up upon VScode opening and Run/Upload commands also do not work. 

I tried disconnecting FiPy and reconnecting it in the attempt of forcing a terminal to pop up where it connects, however, no response was found. 

I've attached two images indicating how the bottom-left Pymakr button is missing and the responses were given when pressing run/upload buttons. Additionally, I've attached how Pymakr is missing from the terminal as an option.

Will there be an update soon in response to the new VScode update?

Thanks

elmasria, 2021-02-05:
Hi @antifm 

Thank you for reporting this. we will fix this issue and upload a new version of Pymakr 

Best Regards,
Ahmad EL Masri

peter-pycom, 2021-02-19:
Fixed version has been released

closed 2021-02-19 by peter-pycom