pycom/pycom-micropython-sigfox #538: Pygate Gpy v1.20.2.r4 run time error

pauldeng, 2021-03-29:
Hi Pycom team,

I can successfully build v1.20.2.r4 from source, but got runtime error.

* Pygate + Gpy
* v1.20.2.r4

I build the r4 from source by:
```bash
make BOARD=GPY BTYPE=release VARIANT=PYGATE RGB_LED=disable MOD_SIGFOX_ENABLED=0 MOD_COAP_ENABLED=0 PYBYTES_ENABLED=0 V=1 flash
```

Then I flash the tar.gz file and run the [Pygate example](https://docs.pycom.io/tutorials/expansionboards/pygate/).

I got run time error and concentrator stopped. Here is the log:
```text
Uploading project (main folder)...
Not safe booting, disabled in settings

Uploading to /flash...
Reading file status
Failed to read project status, uploading all files
[1/3] Writing file boot.py (0kb)
[2/3] Writing file config.json (6kb)
[3/3] Writing file main.py (1kb)
Upload done, resetting board...
OK
Starting LoRaWAN concentrator
Connecting to WiFi...... OK  
Syncing RTC via ntp.... OK 

[2083451660] LORAPF_INFO_:lora_gw_init() start fh=309460 high=8852 LORA_GW_STACK_SIZE=15000
[2083451661] LORAPF_INFO_:lora_gw_init() done fh=294456 high=8852
Pycom MicroPython 1.20.2.r4 [v1.20.1.r2-363-ga37510c09-dirty] on 2021-03-29; GPy with ESP32
Type "help()" for more information.
>>> [2083451675] LORAPF_INFO_:*** Packet Forwarder for Lora PicoCell Gateway ***
Version: undefined
[2083451683] LORAPF_INFO_:lorapf *** Lora concentrator HAL library version info ***
Version: " 0.2.3 ";
[2083451693] LORAPF_INFO_:[main] Little endian host
[2083451741] LORAPF_INFO_:[main] lorawan_public 0, clksrc 1
lgw_board_setconf:593: Note: board configuration; lorawan_public:0, clksrc:1
[2083451743] LORAPF_INFO_:[main] antenna_gain 0 dBi
[2083451750] LORAPF_INFO_:[main] Configuring TX LUT with 16 indexes
[2083451753] LORAPF_INFO_:[main] radio 0 enabled (type SX1257), center frequency 922800000, RSSI offset -163.999996, tx enabled 1
lgw_rxrf_setconf:627: Note: rf_chain 0 configuration; en:1 freq:922800000 rssi_offset:-164.000000 radio_type:2 tx_enable:1
[2083451775] LORAPF_INFO_:[main] radio 1 enabled (type SX1257), center frequency 921800000, RSSI offset -163.999996, tx enabled 0
lgw_rxrf_setconf:627: Note: rf_chain 1 configuration; en:1 freq:921800000 rssi_offset:-164.000000 radio_type:2 tx_enable:0
[2083451797] LORAPF_INFO_:[main] Lora multi-SF channel 0>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-200000 SF_mask:0x7e
[2083451815] LORAPF_INFO_:[main] Lora multi-SF channel 1>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:0 SF_mask:0x7e
[2083451832] LORAPF_INFO_:[main] Lora multi-SF channel 2>  radio 1, IF 200000 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:200000 SF_mask:0x7e
[2083451850] LORAPF_INFO_:[main] Lora multi-SF channel 3>  radio 1, IF 400000 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:400000 SF_mask:0x7e
[2083451867] LORAPF_INFO_:[main] Lora multi-SF channel 4>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-400000 SF_mask:0x7e
[2083451885] LORAPF_INFO_:[main] Lora multi-SF channel 5>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:-200000 SF_mask:0x7e
[2083451903] LORAPF_INFO_:[main] Lora multi-SF channel 6>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:0 SF_mask:0x7e
[2083451920] LORAPF_INFO_:[main] Lora multi-SF channel 7>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
lgw_rxif_setconf:746: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:200000 SF_mask:0x7e
[2083451938] LORAPF_INFO_:[main] Lora std channel> radio 1, IF -400000 Hz, 125000 Hz bw, SF 9
lgw_rxif_setconf:720: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:-400000 bw:3 dr:8
[2083451954] LORAPF_INFO_:[main] FSK channel 8 disabled
lgw_rxif_setconf:655: Note: if_chain 9 disabled
[2083452011] LORAPF_INFO_:[main] gateway MAC address is configured to lX
[2083452012] LORAPF_INFO_:[main] server hostname or IP address is configured to "ns-dev.agersens.systems"
[2083452016] LORAPF_INFO_:[main] upstream port is configured to "1700"
[2083452022] LORAPF_INFO_:[main] downstream port is configured to "1700"
[2083452029] LORAPF_INFO_:[main] downstream keep-alive interval is configured to 10 seconds
[2083452037] LORAPF_INFO_:[main] statistics display interval is configured to 30 seconds
[2083452044] LORAPF_INFO_:[main] upstream PUSH_DATA time-out is configured to 100 ms
[2083452052] LORAPF_INFO_:[main] packets received with a valid CRC will be forwarded
[2083452059] LORAPF_INFO_:[main] packets received with a CRC error will NOT be forwarded
[2083452067] LORAPF_INFO_:[main] packets received with no CRC will NOT be forwarded
[2083452078] LORAPF_INFO_:[main] found global configuration file and parsed correctly
lgw_calibrate_sx125x:512: Note: calibration started (time: 2300 ms)
lgw_calibrate_sx125x:533: Note: calibration finished (status = 0xBF)
lgw_calibrate_sx125x:570: calibration a_i = 62
lgw_calibrate_sx125x:570: calibration a_i = 49
lgw_calibrate_sx125x:570: calibration a_i = 38
lgw_calibrate_sx125x:570: calibration a_i = 34
lgw_calibrate_sx125x:570: calibration a_i = 26
lgw_calibrate_sx125x:570: calibration a_i = 20
lgw_calibrate_sx125x:570: calibration a_i = 16
lgw_calibrate_sx125x:570: calibration a_i = 13
lgw_calibrate_sx125x:512: Note: calibration started (time: 2300 ms)
lgw_calibrate_sx125x:533: Note: calibration finished (status = 0xBF)
lgw_calibrate_sx125x:570: calibration a_i = 33
lgw_calibrate_sx125x:570: calibration a_i = 25
lgw_calibrate_sx125x:570: calibration a_i = 20
lgw_calibrate_sx125x:570: calibration a_i = 16
lgw_calibrate_sx125x:570: calibration a_i = 13
lgw_calibrate_sx125x:570: calibration a_i = 10
lgw_calibrate_sx125x:570: calibration a_i = 8
lgw_calibrate_sx125x:570: calibration a_i = 7
Info: Initialising AGC firmware...
Info: putting back original RADIO_SELECT value
[2083459260] LORAPF_INFO_:[main] concentrator started, packet can now be received
[2083459261] LORAPF_INFO_:[down] st8art4592[20628345]
LORAPF_E2RROR0:[m] a8in] impossible toL c34rea59te Timer Sy] LncORAPF_INFO_:[ueaRd (p  ]2 )PF_IN(FO3_4:[jit ]11 sta,22,1)

start
[208[2083459276]3459276] LO LORAPF_INFRAPF_INFO_:[O_:[up  ] Ejit ] end
nd of upstream thread

[2083469296] LORAPF_INFO_:[down] End of downstream thread

[2083469297] LORAPF_INFO_:[main] Stopping concentrator
```

I also tried the official r4 release and it works fine:
```text
... ...
Info: putting back original RADIO_SELECT value
[2083720970] LORAPF_INFO_:[main] concentrator started, packet can now be received
[2083720971] LORAPLoRa G[W208[2sta0837rt2e[d20
83720F9720972] LO_INFO_:[dow0971] LORAP73] LORAPF_IRAPF_INFO_:n] start
F_INFO_:[up  ]N start
[jit ] stFO_:[ts  ] start
art
[2083750974] LORAPF_INFO_:[main] report
##### 2021-03-29 04:17:34 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### [JIT] ###
[jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
```

I think there could be some configuration I missed and caused this error. Hope the Pycom team can help me out.

Thank you!

pauldeng, 2021-03-29:
Hi Pycom team,

Issue is fixed by re-do the development environment setup. After git clone the source code again. The v1.20.2.r4 built from source start to work as expected.

I hope this message will help others.

Thanks for your support.

Cheers,
Paul

closed 2021-03-29 by pauldeng

peter-pycom, 2021-03-30:
I'm pretty sure the actual problem was that you ran out of memory:

> To avoid running out of memory one should not run the WiFi task and the LTE task at the same time. .....

https://docs.pycom.io/tutorials/expansionboards/pygate/#setup