pycom/pycom-micropython-sigfox #539: can't upgrade firmware on expansion 3.1 with wipy 3.0 no comport available

johncblacker, 2021-03-30:
Please include the following information when submitting a bug report:
wipy 3.0
expansion board 3.1
trying to follow directions, but having much difficulty.
1.) using zadig, I see the libusbK USB device
2.) uninstall device, install new device with libusbK driver...works
3.)plug in usb while holding down s1, I see unknown device but then it goes back to libusbK device
4.)try running dfu util, fails saying no dfu device available
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 04d8
Match product ID from file: ef99
Cannot open DFU device 04d8:ef99
No DFU capable USB device available

5.) put wipy board back onto expansion board and I still see the libusbK device but nothing listed in com ports for board.
Don't know what to do.  When I plug in usb while holding down s1 I don't know if in dfu mode - no lights, only indication is that I have "unknown" device for a short while, then back to libusbK device.  Now I have unusable expansion board and wipy as well.  Seems like this procedure is overly complicated.  What to do?

johncblacker, 2021-03-31:
> Hey,
> I think you set the driver the wrong way around. I know the procedure for windows is somewhat confusing. You're supposed to set the libusbK driver for the dfu-mode (so, in the first 7 seconds after pressing button S1), and reset the driver in normal mode to whatever it normally is (the 'usbser' as shown in the image) https://docs.pycom.io/updatefirmware/expansionboard/. It appears you set the zadig driver for normal mode, instead of the dfu mode
>
> Best,
> Gijs

Well, OK then, I’ll put the winusb driver back on when it’s in “normal” mode and see if I get a com port; but what about the inability to run the dfu-util to load firmware on the expansion

board?

johncblacker, 2021-03-31:
Setting to Winusb.... didn't work, so I changed it to USBSER(CDC) and now I have a com port and I can get to the board with Thonny terminal.  But, what I really wanted to do in the first place is to load the Loboris distribution of micropython onto the board.  I should be able to run it because the board is an ESP32 based processor.  But, I can't get ESPTOOL to do anything with the board.

johncblacker, 2021-03-31:
Well, I think I figured out how to do what I want.  On the pin-out diagram shows two bootloader modes: low (p2 -> GND) and boot modes and safe boot (p12 -> 3.3v).  I tried the latter but all that seems to do is cycle through the 3 different system restores and while it's doing that, ESPTOOL can't get to the board.  However, low level bootloader mode puts the board in a mode that ESPTOOL can connect to and I was able to get to the board and read from it (mac id).  So, now for flashing Loboris!  I thank you for your info as it got me back to the point that I was able to get a com port!  I'll tackle the dfu-util issue another time and with another post.  Thanks again Gijs - much appreciated the rapid response!

johncblacker, 2021-03-31:
Closing, although I wasn't able to get dfu-util to work, I was able to get back to a board that is recognized by host and has a com port.  Now I can try load of Loboris distro.  Putting off dfu-util issue to another day...

closed 2021-03-31 by johncblacker

gijsio, 2021-03-31:
I'll try to make the DFU-Util instructions more clear in general. The tool is only for updating the firmware of the expansionboard (the USB-Serial converter) and has nothing to do with the pycom firmware or esptool.

johncblacker, 2021-03-31:
I do understand that; but I was unable to get the dfu-util to load the firmware that I had downloaded.  The message I got said it couldn’t find an dfu capable devices; but, I didn’t try that

utility when in “low level bootloader” mode (could this be dfu mode)?  Anyway, it would be much more helpful if when one boots while holding down the “s1” button and plugging in the

usb cable a led would flash to indicate that it’s in the correct mode (I didn’t even know how to put it into the “llb” mode until I was looking at the pin-out diagram.  Also, most of the doc

says Gnd -> pin 23, but I believe that should be 3.3v -> pin 12 for the “safe mode” and the “LLB” mode is Gnd -> pin 2.  Much confusing info out there that’s hard to sort through.  But 

thanks again for the response.

 

jb