pycom/pycom-micropython-sigfox #540: Program LOPY4 in C/C++

gdferreira99, 2021-04-05:
Hi,

    I am not an expert in this, sorry if I say something stupid :D
    
    I have a LOPY4 and i want to attach it to a PyGate Expansion Board. I tried to use the firmware pygate version from Pycom but it was limited for me because i want to program de packet forwarder (or trye) without having an LTE or Wi-Fi connection. I want to make a Lora repeater with 8 channels only working by LORAWAN. 
    I checked the scripts and libraries in this repository and I noticed that it has C/C++ code for configuring LOPY4 and also PyGate (sx1308) and they give me a little more room to work on (the pygate firmware of pycom only has a function init that need tcp and rtc synchronization).
    I followed the steps from this repository to install pycom-esp-idf in linux, clone this repository and did make in mpy-cross. I was also able to do make BOARD=Lopy4 clean, make BOARD=Lopy4 and also flash to the microcontroller in the esp32 directory, writing in its memory registers.
    Until here, all seems to work fine. My problem now is, I don't understand what I am flashing to the board when I do make BOARD=LOPY4 flash in the esp32 directory. I do that and nothing seems to happen. Also I cant figure out how to use the functions that are in the files of this repository through a script of my own so I could program the board in C. I tried to use this repository in, for example, PlatformIO, but it gives me many errors because of the includes.
    Basically, I want a way so that I can program the board in C with the material of this repository, can somebody help me??

Thank you very much!

gijsio, 2021-04-09:
No worries,
You're flashing the pycom micropython firmware to your device, which is compiled from C. If you cloned the repository and you're able to build the firmware you're actually almost there. In the `esp32` folder of the repository, you can find most of the code used. To list some examples:
* In the `mods` folder, you can find all micropython modules (like machine, pycom, network etc.)
* The `lte` folder contains all basic LTE functionality etc.
* There is a pygate folder with all the sourcecode for the Pygate (which is probably what you're looking for)
You can open the files as a project folder in your IDE, but I would not recommend using platformIO for the building and flashing, as it can interfere with the esp-idf version used, and will indeed throw some random errors. Not using Platformio might also red-flag some of the includes in the editor, but that will be fixed by the compiler

Note that if you want to compile the pygate functionality into the firmware and flash it, you'll have to use the flag `VARIANT=PYGATE` in the command.

Let me know if that answers your question

gdferreira99, 2021-04-09:
To see if I understand, the firmware is compiled in C, but to use the libraries built in C, I program in micropython?

My goal was to change the functionality of the pygate, i don´t want to use it as a Gateway. I want to built a 8 chanell Lora Repeater, and for that I can´t use TCP, only LoraWAN. The function from the PyGate firmware only starts with a TCP connection, I want to change that. 
I saw that the Legacy firmware is more low-level, but I cant understand with libraries I can use with that firmware. 

In last case, should I edit the C files so that the firmware that i upload do that??

Thanks for your reply man

gijsio, 2021-04-12:
The microcontroller we use (the ESP32) is to be programmed in C using the esp-idf. Using some clever code we can interpret python code using the pycom or micropython firmware. 

In the Pycom ecosystem, we support all functionality of the pycom firmware, though there are always features missing, like the one you're looking for. In some cases, they can be programmed in micropython as well using externally imported modules, requiring no change in the firmware, but in this case, you'll have to change the firmware in order to add the functionality, as by default, the Pygate firmware does not support the repeater functionality. 

You can use legacy firmware, though we no longer support it. Though it does not include, amongst others, the basic pygate functionality (the hardware abstraction layer). I'd recommend you start with the latest pygate release (1.20.2.r4) and work from there, though it might still be a lot of work.

gdferreira99, 2021-05-22:
why is in lgw_receive function, called on the thread_up in the pkt_fwd, has a comment saying "Crashing here"????

gijsn, 2021-05-25:
I believe that is a leftover comment from development and no longer valid

gijsio, 2021-05-25:
I believe that is a leftover comment from development and no longer valid