pycom/pycom-micropython-sigfox #548: Fipy Beta Firmware 1.20.3.b3 BLE advertisement sniffer performances drop

Ichicken32, 2021-05-12:
Hi,

I’m using a Fipy as BLE Central Device to catch advertisement frames of a peripheral device. I used the previous Beta Firmware (v1.20.3.b0). I tried the new one (v1.20.3.b3) and there is lowering in the performances.

To give numbers, the Fipy catched around 85% of the emitted frames with the previous Beta Firmware (v1.20.3.b0) and we fall to 50% with the new one (v1.20.3.b3).

The tests have been done with the same Software, the same board, the same emitter and in similar condition. The advertisement interval was 500ms.

Best regards.

CoreSoftwareTeam, 2021-05-19:
Hi @peter-pycom ,

Any ideas concerning BLE performance ? Or maybe tips to compile a new firmware from the B3 firmware to realize tests and keep you informed ?

Regards,

swuser-ai, 2021-05-19:
Hi @peter-pycom ,

Any ideas concerning BLE performance ? Or maybe tips to compile a new firmware from the B3 firmware to realize tests and keep you informed ?

Regards,