pycom/pycom-micropython-sigfox #552: Pysense 2.0 Light Sensor LTR329ALS01 returning always 0 values after waking up from deepsleep

StefanoOrdine, 2021-05-17:
Please include the following information when submitting a bug report:

* Exact steps to cause this issue
    1. First I did this...
    2. Then, I did that...
* What you expected
    * e.g. I expected *abc* to *xyz*
* What happened instead
    * e.g. Instead, I got *zyx*

Thank you!
```
>>> import os
>>> os.uname()
(sysname='FiPy', nodename='FiPy', release='1.20.2.r4', version='v1.11-ffb0e1c on 2021-01-12', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.6.1')
```
```
from pycoproc_2 import Pycoproc
>>> py = Pycoproc()
>>> print(py.read_fw_version())
16
>>> print(py.read_hw_version())
5
```
Hello,
I am experiencing an issue with Pysense 2.0 Light sensor LTR329ALS01, I am attaching the steps to reproduce the issue:

1. turning on the power
2. running the main.py (added just main parts):
```
import pycom
from pysense import Pysense
from LTR329ALS01 import LTR329ALS01
...
py = Pysense()
lt = LTR329ALS01(py)
light = lt.light()
print(light) # --> (145,23)
...
pybytes.deepsleep(time_milliseconds)
```
3. the FyPy goes to deepsleep
4. the FyPy wakes up and execute again the code at point 1
5. the print returns (0,0)

I was expecting at point 5 a tuple with the new light readings instead of zero values.

gijsio, 2021-05-21:
Hi,
Could you post this issue to the pycom-libraries repository instead? We maintain the sensor libraries over there: https://github.com/pycom/pycom-libraries
Also, it might be worth mentioning the related forum thread you posted in: https://forum.pycom.io/topic/1462/solved-pysense-light-sensor-shows-zero-after-cycling-through-deep-sleep/ 
I will close the issue here

StefanoOrdine, 2021-05-21:
sure, thanks for the suggestion!

closed 2021-05-21 by StefanoOrdine

JayeshEngibrains, 2024-01-22:
@StefanoOrdine 
I am facing a similar issue with the light sensor(LTR-329ALS).
Did you find the root cause?