pycom/pycom-micropython-sigfox #553: Firmware Update Fail, failed to create record

johnmcdnz, 2021-05-18:
I have new(ish) Fipy boards with Pysense expansion boards,

Tried to update the firmware using the updater tool on Windows.  This connects ok on com port.

The update tool proceeds until it starts the flash, see image
[image]

I had the same issue when trying a different expansion board that does work.  I have a Lopy that does flash ok.

I was able to do a manual update using FiPy-1.20.2.r4.tar
But I'm not sure what region settings this results in for the Sigfox and Lora modules.

TIA.

Xykon, 2021-05-18:
Sorry this should be fixed now.

johnmcdnz, 2021-05-19:
Thanks Christian.  The units seem ok on the manual install.  I'll check them out over next few days as we get going.

closed 2021-05-19 by johnmcdnz