pycom/pycom-micropython-sigfox #558: UART communication problem with Lopy and ESP32

iamlao, 2021-06-02:
Hello, 

Here is my material : 
- Pycom  expansion board v2.1A
- Firmware : (sysname='LoPy', nodename='LoPy', release='1.20.2.r4', version='v1.11-ffb0e1c on 2021-01-12', machine='LoPy with ESP32', lorawan='1.0.2', pybytes='1.6.1')

I am trying to make a Lopy and an ESP32 communicate by UART. 
The ESP32 send and The Lopy receive, for both I am using Pymakr. 

First, to be sure that the Lopy can receive, I try to make it communicate with an Arduino and It worked so I assume that the problem comes not from the Lopy.

Then I tried to make the esp32 send to an Arduino and It did not work, I only received '?'.
I tried to make the esp32 communicate with the Lopy but I only receive incoherent values. 

I expect the lopy to receive 320 as a bytes but instead I receive this : 
[image]

Lopy code : 
`uart = UART(1)
uart.init(9600, bits=8, parity=None, stop=1, pins=('P4','P3')) #P4 = TX = G11 & P3 = RX = G24

while True:
    pycom.rgbled(0x0C5A00)  
    #The board is listening
    recv = uart.read(4)
    if recv != None:
        print(recv[0])`

ESP32 code : 
`uart = UART(1, 9600)                 
uart.init(9600, bits=8, parity=None, stop=1)

while True:
    uart.write(bytes(320))
    time.sleep(0.5)
`
Have you already had this ?

robert-hh, 2021-06-03:
I assume you made the proper wiring, including the GND wire. Then, your code is sending a bytes object consisting of 320 all 0x00 bytes. If you want to send the number 320 as bytes object, you have to use e.g. uart.write(b"%d"%320), or if you want to send the three digits 0x02, 0x03, 0x00, then you have to write uart.write(bytes((0x02, 0x03, 0x00))) or uart.write(b"\x02\0x3\x00)").
Note 1: The default assignment for UART1 is P3=TX, P4=RX, but your re-assignment should work as well. I never has any issues with the UART.
Note 2: Questions like this one are better asked in the forum.

iamlao, 2021-06-03:
@robert-hh Thank you for your response :) 

I didn't use a GND wire, I added it.

I changed the pins : uart.init(9600, bits=8, parity=None, stop=1, pins=('P3','P4'))
For the wiring, I connect 3 things between the Pycom and the ESP32:
- GND
- G11 (Pycom) to TX (ESP32)
- G24 (Pycom) to RX (ESP32)

I tried the three ways you gave to send uart and I got : 
Mainly I receive b'\xe0' and somethimes as you can see It is a little chaotic. 
[image]

I tried to convert b'\xe0' with chr() and I got 224.

robert-hh, 2021-06-03:
The mos simple test for UART is always the loopback test. Connect TX and RX and verify, that you can receive what you have sent. That should however work for both boards. Besides that, I would look with a logic analyzer at the wires. A cheap 10€ device is sufficient.

iamlao, 2021-06-03:
@robert-hh 
I tried the loopback test with this code and the esp32 : 
`uart = UART(1, 9600)                 
uart.init(9600, bits=8, parity=None, stop=1)

while True:
    uart.write(bytes(0x01))
    recv = uart.any()
    print(recv)
    time.sleep(0.5)
`
When TX and RX are connected, nothing is displayed but if they are not connected, It displayed a 0.

robert-hh, 2021-06-03:
For the loopback test, the only connection required and allowed is RX to TX.