pycom/pycom-micropython-sigfox #564: Pybytes error at tab "DEVICES"

r3htanz, 2021-09-08:
Please include the following information when submitting a bug report:

* The Pycom board you are using: WiPy 3.0

Not able to load "Devices" tab on pybytes. It gets stuck at the loading screen and results with the following error:
"Network error: Failed to fetch"

elmasria, 2021-09-09:
Hi @r3htanz, 

Thank you for your feedback.

Which Email you have used for Pybytes?  

Best Regards,
Ahmad EL Masri

r3htanz, 2021-09-09:
> Hi @r3htanz, 
> 
> Thank you for your feedback.
> 
> Which Email you have used for Pybytes?  
> 
> Best Regards,
> Ahmad EL Masri

Hi Ahmad,

thanks for reaching out! I don't want to post my email address to the public.
But if this issue only links to my pybytes account is it possible to send you my email address in a private way to solve this problem?

Best regards,
Jan

elmasria, 2021-09-09:
HI @r3htanz 

Yes sure, you can reach me at (ahmad@pycom.io)

Yes most probably is related to your account. let's talk via email.

Have a nice day.

Best Regards,
Ahmad EL Masri

elmasria, 2021-09-13:
Hi @r3htanz,

Do you still face this issue?

Best Regards,
Ahmad EL Masri