pycom/pycom-micropython-sigfox #566: can't add vodafone SIM card to pybytes

ozembuch, 2021-09-21:
I order vodafone SIM card via pycom eshop. After adding Order number, it should add SIM card to pybytes. But even the message says SIM card was added, I do not see any SIM cards fro manage. Without this step I cant activate SIM card
c1dfb4fa-8da5-40df-a6c8-716ebb195069-image.png

elmasria, 2021-09-21:
Hi @ozembuch,

Thank you for reporting this, 

We are working with the third party (Vodafone) to fix this issue and will update Pybytes ASP. 

Have a nice day,

Best Regards,
Ahmad EL Masri

razvandragomirescu, 2021-09-21:
I have the exact same issue and it is blocking us from using purchased SIMs, anxious to get this solved.

razvandragomirescu, 2021-10-03:
Is this fixable or should I ask for a refund? The cards are currently unusable since they are not activated and the clock is ticking, we've already lost 1 month out of the 12 we paid for.