pycom/pycom-micropython-sigfox #583: unable to communicate to pybytes app

Inzemon, 2021-12-02:
Please include the following information when submitting a bug report:

* The Pycom board you are using:  LoPy4 + Exp. board 3.1
* The firmware version you are "attempting" to use :  LoPy4-1.20.2.r6             from pycom_firmware_updater_1.16.5
* Exact steps to cause this issue
    1. First I registered to pybytes : https://pycom.io/products/software/pybytes-3/
    2. Then, as requested I updatedwith "ONLINE FIRWARE UPDATER"
* What you expected on Windows 10, using ATOM 1.54 and Pymakr 2.2.0
    * was to connect properly to my LoPy4
* What happened instead
    * I got : "Connecting to COM5... " on ATOM/Pymakr
* After Reset on LoPy4 board I got the following : 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2128
load:0x4009fa00,len:19760
entry 0x400a05bc

* Before this update I was able to upload folders and run micropython scripts.

Thank you for your help

Ztubben, 2021-12-06:
Got the same issue with FiPy + exp board 3.1. I have connected several units with pybytes before but now it is impossible. Downgrading firmware on the pycom board does not work.

Xykon, 2021-12-07:
Have you tried pushing the safe boot button when resetting the device? Are you able to use the erase_all option with the CLI firmware updater?

Ztubben, 2021-12-07:
Yes I have. Unfortunately none of those methods work.

Inzemon, 2021-12-07:
Agree, same for me.

Inzemon, 2021-12-07:
I guess if I was able to (re)install the previous version, it'll be fine. But The version I got (LoPy4-1.20.3.b3.tar.gz)  is not that easy to install. Do you have some "How-To" you can share?

Inzemon, 2021-12-09:
I have also send a request (including the log file) to Pycom and 7 days later no reply but a confirmation request.

Inzemon, 2021-12-09:
After reviewing the documentation, I missed up the update firmware for the expansion board. I did it successfully, then updated again the Device "Lopy4", to finally got the same issue on Atom with Pymark :
"...Connecting to COM5..." 
Safe boot -> no effect
Reset got same log as previous
The Main LED is blinking BLUE
The USB LED is Green
The "LED" LED is 50% bright orange. 

So no access to my USB port even after these updates. I'm stuck, as I can't enter python scripts.