pycom/pycom-micropython-sigfox #588: machine.sleep(1) unstable

stevegoobermanhill, 2022-02-02:
**WiPy 3.0**

sysname='WiPy', 
nodename='WiPy',
release='1.20.2.r6', 
version='v1.11-c5a0a97 on 2021-10-28', 
machine='WiPy with ESP32', 
pybytes='1.7.1')

Hi. 
machine.sleep(1) appears to be causing core dumps. 

Sample code - switching machine.sleep() to time.sleep_ms() removes the issue but obviously has higher power consumption:

```
#setup ADC 
adc = ADC(0)
adc_c = adc.channel(pin = ADC_PIN, attn=ADC.ATTN_2_5DB) 

while (True):
    v = adc_c.value()

    if (v >= THRESHOLD):
        #do_something

    machine.sleep(1)
```
Error message and core dump as follows:

Thxs, Steve

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094d2f  PS      : 0x00060e33  A0      : 0x80087d60  A1      : 0x3ffb7130  
A2      : 0x00000000  A3      : 0x00060e20  A4      : 0x00000001  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x3ffaff34  A8      : 0x0000abab  A9      : 0x3ffb7130  
A10     : 0x00000003  A11     : 0x0000cdcd  A12     : 0x00060e20  A13     : 0x0003d0f4  
A14     : 0x3f800010  A15     : 0x3f9400d1  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40092834  LEND    : 0x40092862  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x40094d2f:0x3ffb7130 0x40087d5d:0x3ffb7160 0x40085b35:0x3ffb7180 0x40136ca6:0x3ffb71c0 0x400d23fc:0x3ffb71e0

================= CORE DUMP START =================
7CsAAAEAAAANAAAAbAEAAA==
fHL7P3Bw+z90cvs/
4HD7PxBy+z/ydB70vEH8P7xB/D98cvs/tEH8PwMAAAAgPvw/ID78P3xy+z8AAAAA
FgAAAHhm+z9lc3BfdGltZXIAwUQWnYQAAAAAAHRy+z8AAAAAIAwGABYAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
WIZAPwAAAABYNwlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABXgSA==
AQAAAC9NCUAzDgYAYH0IgDBx+z8AAAAAIA4GAAEAAADNzQAA//8/szT/+j+rqwAA
MHH7PwMAAADNzQAAIA4GAPTQAwAQAIA/0QCUPwAAAAAdAAAAAAAAADQoCUBiKAlA
/////yAOBgD00AMAADIIQNQE9z8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAIBx+z8AAAAA/////0Dd+z84WwiAYHH7PwAAAAABAAAA
1AT3PwAAAAA0KAlAYigJQP////9kNAhA8EH8P1hmCUCpbBOAgHH7P2AE/D/AAJQ/
GAAAAAAYAAADAAAAIwAGAP8jDYDAcfs/AQAAAMAAlD+lFAiAoHH7P+zc+z8kDQAA
CgAAAMAAlD8BAAAADHwIQAoAAADwBvw//gAAAAJwAAAAAAAA4HH7PwEAAABQPvw/
YAT8PwAAAAAkcPQ/AAAAAAAAAAAQcvs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAABQPvw/hGwTQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABxy+z8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
zEP8P8Bu/T+4c/0/
wG79P1Bz/T/KhQAA4ED8P+BA/D/MQ/w/2ED8Pw4AAAA4Rfs/OEX7P8xD/D8AAAAA
CwAAALxD/T9NaWNyb1B5AAAAAAAAAAAAAQAAALhz/T8BAAAAIAcGAAsAAAACAAAA
[... 210 lines cut ...]
AAAAABBD/T8AAAAAAAAAAAMAAAAwQvs/AQAAAP4G/D///z+zAAAAAAAAAAAAAAAA
1WcSgFBC+z8CMAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAcQ/0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
JKX9P4Cj/T8cpf0/
gKP9P7Ck/T9k55ceQKv9P4A//D8kpf0/eD/8PxQAAABn3DMjC6JN2iSl/T8AAAAA
BQAAACCV/T9Tb2NrZXQgT3BlcmF0aW8AAQAAAByl/T8AAAAAIQAGAAUAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
WIZAPwAAAABYNwlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAAC0cFA==
ZDIIQPC/AEAwAAYA9E0JgECk/T8AAAAAIwAGACAABgAQcv0/VQAAACCD/T///z+z
q6sAABDd+z8jAAYAJKX9PwEAAAD+AAAAAAAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAGQ0CEABAAAAWGYJQHQ3+T8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD6XgmAUKT9P0Dd+z8GAAAA
ulwOgHCk/T8BAAAA9EH8PySl/T8BAAAA/gAAALCd+z8AAAAAkKT9PwAAAAAAAAAA
8EH8P0Dd+z9kP/w/aED8PwAAAACwpP0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAvKT9PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
================= CORE DUMP END =================
```

msariisik, 2022-03-01:
Hi @stevegoobermanhill, 

Thank you for your detailed report.

I followed the same steps, but I couldn't reproduce your issue. Can you provide us with which pin you are using to read ADC value?

stevegoobermanhill, 2022-03-09:
> You don't often get email from ***@***.*** Learn why this is important<http://aka.ms/LearnAboutSenderIdentification>
> PHISHING ALERT
> This email originated from outside of Amey
> Be aware that this could be a phishing attempt, do not click links or open attachments unless you recognise the sender and know the content is safe.

Hi Meric,
thanks for getting back to me.

Pin 13

Steve

Amey plc is a company registered in England and Wales. Registered Office: Chancery Exchange, 10 Furnival Street, London, EC4A 1AB. Registered Number: 4736639. For particulars of companies within the Amey Group, please visit http://www.amey.co.uk/Home/Companyparticulars/tabid/182/Default.aspx.

This email and accompanying attachments may contain confidential or privileged information and are for the sole use of the intended recipient(s). If you are not the intended recipient, you are notified that any use, dissemination, distribution or copying of this email or accompanying attachments is prohibited. If you received this email in error, please notify us immediately and delete all copies of this email and accompanying attachments.

Email does not guarantee the confidentiality, completeness or proper receipt of the messages sent and is susceptible to alteration.

Please note that Amey monitors incoming and outgoing emails for compliance with its Security Policies. This includes scanning emails for computer viruses.