pycom/pycom-micropython-sigfox #589: BLE services limited to 3

robgar2001, 2022-02-10:
* The Pycom board you are using.
  WiPy
 
* (sysname='WiPy', nodename='WiPy', release='1.20.2.r6', version='v1.11-c5a0a97 on 2021-10-28', machine='WiPy with ESP32')

* Exact steps to cause this issue
    1. The services we created a ble gatt server side:
Service A : Contains 11 characteristics
Service B : Contains 9 characteristics
Services C, D, E, F : Each contain 5 characteristics

    2. We push values into each characteristic
    3. We look with a ble analyser at the services and characteristics
    
* Wat we expected:
   To see all values updated when the ble analyser app reads them out
* Wat happened isntead:
  Services A, B and D show the correct values, the other ones are all zero. They show up, but just all 
  zero when we read them out.

Thank you!

msariisik, 2022-02-28:
Hi @robgar2001 ,

I tried to reproduce your problem. Are you stating the number of characteristics with nbr_chars argument while creating service?
`srv1 = bluetooth.service(uuid=srv1_uuid, isprimary=True, nbr_chars=11)`

If this is not the missing part, can you provide your basic code to reproduce the problem?

robgar2001, 2022-03-12:
Hi @msariisik ,

Thanks for the response! We located the issue. It was probably due to a strange memory problem with dictionaries. All our characteristics where present, the default values could be read. The new values could also be read server-side when changed.  When we realised this, we starting thinking of a memory problem.

All services and characteristics are kept in a dictionary when the BLE object in them. The dictionary part for service A and B are created beforehand, the parts for services C,D,E,F are added dynamically. This was done by assigning parts of a template. The template was kept as a class variable. 

When we created a new object instead of assigning a part of the template, it worked fine.

If you are still intresseted to find out why, I am willing to create a reproduceable script.

Kind regards,
robgar2001

closed 2022-07-15 by robgar2001