pycom/pycom-micropython-sigfox #592: Problem with a thread freezing the other threads

elientumbaluanga, 2022-04-19:
* Chip : W01
* Firmware : Wipy 1.20.2.r6

Hello ,  I have a program with 4 threads.
One of the four threads is listening to an input.
Following the state of the input[pin17=Pin("P17", mode=Pin.IN)] , i have all of my program not responding.
I did a downgrade of the firmware, and the problem was fixed.
Can you see this problem 
Cordially.

elientumbaluanga, 2022-04-21:
@martijnthe @amotl @ftylitak @DvdGiessen

amotl, 2022-04-28:
Dear Elie,

thanks for pinging me. However, I don't use Pycom MicroPython actively anymore, so apologies that I can't help here. Good luck that your issue will be resolved in one way or another.

With kind regards,
Andreas.

elientumbaluanga, 2022-05-02:
@martijnthe @ftylitak @DvdGiessen