pycom/pycom-micropython-sigfox #593: Is my Lopy4 dead ?

pjuhen, 2022-05-03:
Hi, 

I have a Lopy4 on an expansion board v3.1.

Up to yesterday, I used it connected to my PC.

Yesterday, I plugged a smartphone external battery instead of the PC, to run it oudoor.

Since then the orange LED doesn't light, and the Lopy4 seems dead.

The expansion board is OK, I was able to do a firmware upgrade.

But I have no echo on the REPL.

When I try a firmware upgrade, here is the result :

> sudo ./pycom-fwtool-cli --debug -p /dev/ttyACM0 -s 115200 erase_all
> Running in PIC mode
> Erasing the board can take up to 40 seconds.
> Connecting to ESP32 with baudrate: 115200
> Product ID: 152 HW Version: 7 FW Version: 0.0.11
> Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
> Exception: Failed to connect to ESP32: Timed out waiting for packet header

Is my chip dead ?

Thanks,

Regards,

robert-hh, 2022-05-03:
How did you connect your smartphone battery to the LoPy4?

pjuhen, 2022-05-03:
With an USB cable through the expansion board.

robert-hh, 2022-05-03:
That sounds safe. Do you have a voltmeter to check, if the supply voltages 5V and 3.3V are present at the LoPy?

pjuhen, 2022-05-03:
It seems I have 5V on Vin but nothing on 3.3V.

Thanks,

Regards

robert-hh, 2022-05-03:
Then, the 3.3v regulator is broken, or there is a short in the 3.3V line to GND, either at the LoPy4 or the expansion board. Broken regulators used to happen in the past only when one tried to feed the board through the 3.3V pin (misleading doc), or when the input voltage is too high. Otherwise the regulator is pretty robust and survives shorts and overload.
Replacing the regulator is almost impossible. When that happened in the past, people removed the regulator and supplied the board with 3.3V. But for most people that is no way to go.

pjuhen, 2022-05-03:
I would say the regulator is dead.

I didn't find my voltmeter, but I did a small controller with a LED and 
a resistor.

3.3 is OK (lights the LED) on an ESP32 and an Arduino, but not on the Lopy4.

So, seems that my external battery delivered something that killed the 
regulator.

Is it the expansion board or the Lopy4 ?

Thanks,

Regards,

robert-hh, 2022-05-03:
The regulator is on the LoPy4 board. If there is no signal at the 3.3V pin while sitting on the expansion board, the reason could be:
a) broken regulator on the board. The regulator dies at 7V input voltage or when reverse voltage > 3.3 V is applied to the output.
b) a short to GND either on the LoPy4 or the expansion board.
Choice a) is more likely. Choice b) could be tested with a Ohmmeter.

pjuhen, 2022-05-03:
I did nothing special exepct pluging a powerbank. Maybe this one has 
delivered a peak (it's old).

I ordred a multimeter. More tomorrow...

Thanks,

pjuhen, 2022-05-18:
I think the powerbank was broken and sent a power pulse that killed the regulator. After that, both powerbank and lopy4 were dead.