pycom/pycom-micropython-sigfox #595: Pymesh online software update doesn't work after site update

Ro0t-set, 2022-05-15:
board: FiPy + Pysens

(sysname='FiPy', nodename='FiPy', release='1.20.2.r6', version='v1.11-c5a0a97 on 2021-10-28', machine='FiPy with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.7.1')

Normally after the Pymesh firmware was updated it became 1.20.2.r11, but now it didn't change 

Update Output: 
```
Requesting: manifest.json?current_ver=1.20.2.r6&sysname=FiPy&wmac=f008d1ccdd9c&ota_slot=0x210000 to software.pycom.io:443 with SSL? True
Wrapping socket
Sending request
Requesting: downloads/appimg/firmware_FiPy_1.20.2.r6-pybytes.bin to software.pycom.io:443 with SSL? True
Wrapping socket
Sending request
Full Update Image detected. Restart the device to load the new firmware.
ets Jun  8 2016 00:22:57

```

Even after the update all the pymesh modules do not work

wdebbaut, 2022-05-31:
[image]
As you can see above, I experience the same issue of not being able to update online the pymesh firmware to a connected device on PyBytes.
I wonder if ever this feature worked before. Is there any offline possibility to upload the firmware in my IDE?

Ro0t-set, 2022-05-31:
yes, you can use guide : [doc link]( https://docs.pycom.io/updatefirmware/ota/)
firmware :  [Download link](https://software.pycom.io/downloads/appimg/firmware_FiPy_1.20.2.rc11-pymesh.bin)

however, this will not automatically generate pymesh_config.json

wdebbaut, 2022-06-02:
Thank you sir,
Do I have to flash this bin file with the cli firmware updater? If so, to what partition?

Ro0t-set, 2022-06-02:
I first of all format and install the firmware using the appropriate tool (pycom firmware update), then I do the ota update as explained in the documentaiozne

wdebbaut, 2022-06-04:
I am a bit lost now Tommaso!
With the offline pycom firmware update tool version 1.16.6 (see below, I can only install firmwares with the tar or gz exension. The one you gave me is with a bin extension. How do you format firmware anyways?
[image]

I tried once again the online firmware update for the deployment of the pymesh firmware. Still no result after my first attempt four days ago. I would have hoped someone at PyCom is working on it as we have 20 devices lying around here to be provsioned with pymesh....

Ro0t-set, 2022-06-04:
I try to explain you by points, but the process I found is quite long

- Computer:

  Run this script on your computer (server) this script:  [server script](https://github.com/pycom/pycom-libraries/blob/master/examples/OTA/OTA_server.py)
  to use it, use the official guide:[guide]( https://docs.pycom.io/updatefirmware/ota/)
  Use this as a firmware on server :  [firmware](https://software.pycom.io/downloads/appimg/firmware_FiPy_1.20.2.rc11-pymesh.bin)
  Once the server has been successfully launched on your computer,

- Device:

  1. Install the latest version of the software available through the offline tool
  2.  Connect the device to the same wifi network as your computer
  3. Upload this script via atom to the device (client)[cliente script]( https://github.com/pycom/pycom-libraries/blob/master/examples/OTA/1.0.0/flash/lib/OTA.py)
  4.  Run the script in step 3
  5.  Create the WiFiOTA object example: ` ota = WiFiOTA (ssid, password, ip, port)`
  6.  Run: ` ota.connect (), ota.update ()`

This process is very inconvenient, but it works. Unfortunately it doesn't generate the paymesh configuration file, I found a way to do that too, but if you have old files I recommend using those.