pycom/pycom-micropython-sigfox #600: Project Status

scottbasgaard, 2022-06-07:
It would be great to get an update on the current status of this project from someone.

We bought into Pycom a few years ago and it seems like there's been a shift in focus these last few months/years and there is almost no activity into getting updates here from the core team.

It's a great product and we really believe in it but with so little activity on bug fixes, open issues on this repo, old versions of Micropython, etc. we're eagerly wondering what's next.

scottbasgaard, 2022-06-07:
Literally got this email 5 minutes after posting this:

>**Announcing the “End of Life” of our current portfolio**
Since we designed, developed, and launched our existing range of products from 2015 onwards, a number of components used have become / are becoming obsolete. This, in addition to the current complexities with component sourcing means that it is no longer viable to maintain the existing portfolio. It is therefore our intent to phase out the following product range (powered by the ESP32) by December 2022:
WiPy3
LoPy4
FiPy
GPy
L01
L04
G01
W01

robert-hh, 2022-06-07:
Since there was no product support for a while by Pycom for the existing product range, and a few "hurrah" twitters for new products, that's what I expected. It will be interesting what Pycom will sell in 2023. Since the stop the current products end of 2022, and given the current announcement vs. reality history, the new products announced for Q4/2022 will not be available before 2024.