pycom/pycom-micropython-sigfox #613: Rounding integers with custom decimals throws NotImplementedError

William04A, 2023-03-09:
Hi! I've noticed a bug related to the rounding function.

> **Heads up:** I'm quite confident that this is the same bug as described here: https://github.com/adafruit/circuitpython/issues/1890.

**Board/system information:**

```
(sysname='GPy', nodename='GPy', release='1.20.2.r6', version='v1.11-c5a0a97 on 2021-10-28', machine='GPy with ESP32', pybytes='1.7.1')
```

**Bug report**

Rounding integer values with a custom number of decimals seems to throw a `NotImplementedError`:

```python
>>> round(1, 2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError:
```

If the values is converted to `float`, it works fine:

```python
>>> round(float(1), 2)
1.0
```

robert-hh, 2023-03-09:
Micropython is not CPython, just a small subset. Thus there are many properties which are not implemented. And you seem to have found one. 
I would not call it a bug. It's by intention not available.

William04A, 2023-03-09:
@robert-hh Fair point. Adafruit's CircuitPython has a bug report about this which I linked in the OP which hints that the function is implemented in Micropython, it is just a flag that you enable. It does add some extra code size according to them, but ~100 bytes. Since it is implemented in Micropython, Adafruit considers it a bug and not having the feature presumes that everything you pass to `round()` with the `n` argument is a `float` makes me personally consider this something that should be present in the source code. However, I'm not a MicroPython source code expert of any kind and I know you're much more skilled in that department from conversations on the PyCom forum :)

robert-hh, 2023-03-09:
Yes, the flag is called `MICROPY_PY_BUILTINS_ROUND_INT`, and it is disabled in py/mpconfig.h. If you need it, you have to build the firmware yourself. In MicroPython mainline, it is usually disabled for ports with small flash, and enabled for ports with larger flash. About Pycom: I do not know if there is activity, now that Pycom Inc got bancrupt and some of their business is continued by Pycom BV. Even if the stopped most of their actual products, they will most likely keep MicroPython for their new products. But even then: I could never tell if raising issues in their public repository was noticed. I hardly ever got feedback from Pycom when I did.
P.S.: round(int,int) is kind of a no-op. since it just returns the first int argument.