pycom/pycom-micropython-sigfox #616: Deepsleep error when working with LTE module

VincBont, 2023-09-07:
* Working on Gpy with Pyscan dev board

* Firmware version : (sysname='GPy', nodename='GPy', release='1.20.2.r6', version='v1.11-c5a0a97 on 2021-10-28', machine='GPy with ESP32', pybytes='1.7.1')

I try to run a simple script where I initialize the LTE module, attached it to the network, connect it to the data stream, disconnect it from this data stream, detach the LTE module from the network and finally send the board to a deep sleep. The deep sleep is working well without the LTE code and the LTE code is running correctly. But as soon as I perform LTE action before a deep sleep, the board crash and reboot.

My code:
```import time
import machine
from network import LTE

_lte = LTE()

def attach_lte(lte):
    if not lte.isattached():
        print("Attaching to LTE network ", end='')
        lte.attach(apn="sunrise.internet", type=LTE.IPV4V6)
        while(True):
            if lte.isattached():
                print(" OK")
                break
            print('.', end='')
            time.sleep(1)
    if not lte.isconnected():
        print("Connecting on LTE network ", end='')
        lte.connect()
        while(True):
            if lte.isconnected():
                print(" OK")
                break
            print('.', end='')
            time.sleep(1)

try:
    _lte.reset() # reset LTE module
    print("OK")
    time.sleep(1)
    attach_lte(_lte) # attach LTE module to the network and connect to data stream
    print("LTE work")
    _lte.disconnect() # disconnect LTE module from the data stream 
    print("OK")
    time.sleep(1)
    print("Detaching LTE ... ", end='')  
    _lte.dettach() # detach LTE module from network
    print("OK")
    time.sleep(5)
    print("Sleep mode...")
    machine.deepsleep(1000*10) # enter deepsleep mode for 10 sec
finally:
    print("Finished")
```

Complete output logs:
```rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2128
load:0x4009fa00,len:19760
entry 0x400a05bc
Smart Provisioning started in the background
See https://docs.pycom.io/smart for details 
UART init...
Init...
OK
Attaching to LTE network ........ OK
Connecting on LTE network  OK
LTE work
OK
[... 304 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
TL/9P7C9/T9Ev/0/
sL39P+C+/T8No5KraMX9P7A//D9Mv/0/qD/8PxQAAACvjAGXeewTTEy//T8AAAAA
BQAAAEiv/T9Tb2NrZXQgT3BlcmF0aW8AAQAAAES//T8AAAAAIQAGAAUAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADUZPs/PGX7P6Rl+z8AAAAAAAAAAAEAAAAAAAAA
soZAPwAAAACENwlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAABAAAAALrjZQ==
bDIIQPC/AEAwAAYAIE4JgHC+/T8AAAAAIwAGACAABgBAjP0/VQAAAFCd/T///z+z
q6sAABDd+z8jAAYATL/9PwEAAAD+AAAAAAAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAGw0CEABAAAAWGcJQMzW9z8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAD6XwmAgL79P0Dd+z8GAAAA
Ml0OgKC+/T8BAAAAJEL8P0y//T8BAAAA/gAAALCd+z8AAAAAwL79PwAAAAAAAAAA
IEL8P0Dd+z+UP/w/mED8PwAAAADgvv0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA7L79PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
================= CORE DUMP END =================
Rebooting...
```
I do not understand this behavior, a bit of help will be appreciated.
Thank you!

jonnerd154, 2023-09-07:
LTE.dettach() is non-blocking, but takes a significant amount of time. Try adding a polling/wait loop after your call to dettach() and before the sleep call, that checks LTE.isattached(), and breaks out of the loop when it's false.

jonnerd154, 2023-09-07:
Follow up: I think you also need to call LTE.deinit() after the detach. 

Quick example of the whole thing:
```
# Detach with state check
if _lte.isattached():
    detachTimeoutN = 0
    _lte.dettach()
    while detachTimeoutN < 10:
        if not _lte.isattached(): 
            break

        utime.sleep_ms(1000)
        detachTimeoutN += 1

# de-initialize the LTE module
_lte.deinit()
```

VincBont, 2023-09-11:
That was a good idea but sadly the error still persists even though I correctly detach and deinit the module.