pycom/pycom-micropython-sigfox #617: Fipy not able to connect to device

satsokn, 2023-11-05:
I am using FiPy with pysense board. The development is made on Visual studio code. I programmed the device but after the software was uploaded I cant connect with the device. The code that I uploaded is running since I see the led change colour. Within the terminal I can connect but I cant inteact with the device. I see the reset messages on terminal. 

[image]

How can I erase the flash and reprogramm the device ?

robert-hh, 2023-11-05:
It seems that you have placed your code in main.py and the code disables Ctrl-C for interruption. You can always perform a safe boot by connection the P12 pin to 3.3V and push reset. Then boot.py and main.py are not executed, and you get a REPL prompt.

satsokn, 2023-11-05:
manage to connect to terminal but still cant load device explorer. 
[image]