pycom/pycom-micropython-sigfox #618: How can I factory reset device (Fipy) ?

satsokn, 2023-11-07:
I cant access the device explorer so I cant write code on main. Is there any way to erase flash and return device to factory settings with empty main?

robert-hh, 2023-11-07:
You can always use esptool.py to erase the flash completely, and then use the pycom updater to reinstall the firmware. But that is quite risky, given that Pycom INC does not exist anymore any I do not know if the update servers still run.
If you just want get rid of main.py, connect Pin P12 to 3.3v and push reset. Then the device boots and skips main.py. Using a simple serial terminal like Putty you can then access the REPL. There, execute:
```
import os
os.remove("main.py")
```

satsokn, 2023-11-09:
when I try to erase main I get the following error
[image]

I have access on terminal and the device response normaly on commands but I cant access main. When I try to open the device in file explorer VS stucks on load

robert-hh, 2023-11-09:
That looks really bad. There must not be a `Guru Meditation Error` when trying to remove a file. I guess that:
```
import os
print(os.listdir("")) 
```
results in an error as well. At least, the file system is corrupted. You could try to reformat the file system with:
```
import os
os.fsformat("/flash")
```
If that does not help, you have to try a full erase of the flash and reinstall firmware and file system. If that fails as well, even if Pycom servuces are available, then the board may be broken.